oss-slu/Pi4Micronaut #41: 7 Segment Display Support

GreihMurray, 2023-04-06:
Includes the need for a circuit, breadboard diagram, and functioning code to integrate the component with Micronaut

7 Segment display is an I2C device, may be able to leverage included CrowPi library to expedite the process

yrlmanoharreddy, 2024-09-09:
This issue is for documenting the seven segment display #248  

and Implementing Code for 7 Segment Display Integration, which includes writing and testing the necessary code to control the display.

yrlmanoharreddy, 2024-10-21:
Below are the @GreihMurray comments

Includes the need for a circuit, breadboard diagram, and functioning code to integrate the component with Micronaut

7 Segment display is an I2C device, may be able to leverage included CrowPi library to expedite the process