oss-slu/Pi4Micronaut #45: Fan Support

GreihMurray, 2023-04-06:
Includes the need for a circuit, breadboard diagram, and functioning code to integrate the component with Micronaut

Will likely be a PWM device but not certain.

yrlmanoharreddy, 2024-09-23:
Add support for controlling a fan using the Pi4J library. This involves creating a class that handles fan initialization, speed control, and state management.
**Tasks:**
    

    1.  Include a circuit diagram and breadboard layout.
    2. Integrate GPIO pin configuration for fan control.
    3. Create FanController.java class.
    4. Implement methods for:
         Starting the fan
         Stopping the fan
         Setting the speed of the fan
    5. Provide functioning code to integrate the component with Micronaut.

**Acceptance Criteria:** 

      1.     FanController.java class with methods: start(), stop(), and setSpeed(int speed).
      2.     GPIO pin configuration.
      5.     Component operates correctly on hardware.

yrlmanoharreddy, 2024-09-29:
Greih Murray old comments about the issue

Includes the need for a circuit, breadboard diagram, and functioning code to integrate the component with Micronaut

Will likely be a PWM device but not certain.

PR #259 (2024-10-06): 45 Fan Support Helper Tests
Fixes #45

Updated the Gradle build configuration to include support for Pi4J Micronaut. Added the fan control feature with support for turning on, turning off, and setting the fan speed using PWM.

Created FanController.java to provide HTTP endpoints for fan control and FanHelper.java to handle the logic for fan operations, including starting, stopping, and adjusting speed.

Implemented methods: start() to start the fan at full speed, stop() to stop the fan, and setSpeed(int speed) to set the fan speed using PWM with a value range of 0 to 1024.
closed 2024-10-06 by yrlmanoharreddy