oss-slu/Pi4Micronaut #74: Documentation for RGB LED Setup

ruthvikm, 2023-10-05:
Document the circuit process of RGB LED setup like what components are used (with details), pin configuration (clearly label the pins on the RGB LED and other components). Use pictures and circuit diagrams.

kate-holdener, 2023-10-05:
There is a starting point for this documentation: https://oss-slu.github.io/Pi4Micronaut/#_rgb_led
However, it would be nice to have a hardware diagram demonstrating the hardware setup. Also, I see mentions of "unresolved directive" on that page (see attached screenshot), which are probably documentation bugs.
[image]

ruthvikm, 2023-10-05:
@joefol - Here I'm attaching the Circuit Diagram.
Following link we will be useful for your documentation
Link - https://docs.sunfounder.com/projects/raphael-kit/en/latest/c/1.1.2_rgb_led_c.html#c

joefol, 2023-10-22:
Closing this issue, see PR #79 for code changes. Issue created to address the "unresolved directive" bug.

closed 2023-10-22 by joefol