oss-slu/Pi4Micronaut #116: Need to fix LCD1602 Helper

ruthvikm, 2023-11-15:
writeText and writeTextAtLine functions are not working as expected, need to add cursor movement functions too

ruthvikm, 2024-02-07:
Use @Post to make it work instead of @Get

SwindleA, 2024-02-18:
@post versus @get is not the issue. The LCD was working on my version of the OS, but when I try anyone else's version of Pi OS, the LCD does not work. I wiped my OS and tried to replicate how I configured the system and now the LCD does not work for me now. I think at one point I was messing with the paths for the libraries but I do not remember what I did or if that is the actual fix.  

Below is the image of the error. I tried different parameters for the LcdDisplay in the helper. I also made sure to pass the correct number of rows to the LcdDisplay. 

[image]

SwindleA, 2024-03-08:
> @post versus @get is not the issue. The LCD was working on my version of the OS, but when I try anyone else's version of Pi OS, the LCD does not work. I wiped my OS and tried to replicate how I [...]

This was solved by using the 32 bit version the Raspberry Pi Operating System.

ruthvikm, 2024-03-14:
Most of the required functionalities are working

closed 2024-03-14 by ruthvikm