oss-slu/Pi4Micronaut #244: Speed Sensor Support

ruthvikm, 2024-07-29:
- [ ]  Component Description - what it can do
- [ ]  Add yaml configuration
- [ ]  Circuit setup image
- [ ]  Schematic diagram
- [ ]  Instructions on how to use

yrlmanoharreddy, 2024-09-23:
This component will enable the Pi4Micronaut project to measure and monitor the speed of various objects, providing real-time data for control and automation tasks.

**Tasks:**
 1. Provide a circuit setup image for hardware connections.
 2. Create a schematic diagram to illustrate the wiring.
 3. Develop functionality to read data from the speed sensor.
 4. Create a YAML configuration file for sensor settings.

**Acceptance Criteria:**

   1.  Functionality to read and report speed data from the sensor.
   2. YAML configuration file is created and functional.
   3. Circuit setup image is provided and clearly labeled.
   4. Schematic diagram accurately depicts the wiring.
   5. Instructions for usage are clear and easy to follow.
**Expected Outcome:** 
A fully functional speed sensor integration that allows for easy configuration and monitoring.

yrlmanoharreddy, 2024-09-29:
Ruthvik old comments about the issue
- [ ]  Component Description - what it can do
- [ ]  Add yaml configuration
- [ ]  Circuit setup image
- [ ]  Schematic diagram
- [ ]  Instructions on how to use

PR #258 (2024-10-07): Speed Sensor Support.
Fixes #244 

Created Functionality through Helper, Controller files. As well as created documentation for speed sensor by creating an adoc file and a html file. In order to create functionality for a speed sensor, these changed must be made. 

SpeedSensorHelper.java, SpeedSensorController.java, speedSensor.adoc, SpeedSensorHelper.html were created to control the functionality and documentation of a speed sensor. 

Within SpeedSensorHelper.java functions are created in order to handle the requested functionality. to read speed data and convert it to Rotations/Minute(RPMs)  [... cut]
closed 2024-10-07 by yrlmanoharreddy