oss-slu/Pi4Micronaut #345: Good first issue

RalphTan37, 2025-04-14:
**Description:**
Write unit tests for the ActiveBuzzHelper class to ensure the proper functionality of all its methods, including turning the buzzer on and off, executing a beep sequence, playing intermittent tones, and generating Morse code patterns.

**Tasks:**
1) Test Buzzer Methods
Write tests to verify that activeBuzzerOn() correctly sets the duty cycle and frequency, updates actBuzzCheck flag, and logs correct messages		
Write tests for activeBuzzerOff() to confirm that the buzzer turns off and the state flag is updated

2) Test Beep Method
Write a test for the beep() method to simulate the 2-second beep sequence
 so the buzzer state transitions are correct

3) Test Intermittent Tone Method
Write tests for intermittentTone() to validate that the buzzer toggles on and off for the correct durations

4) Test Morse Code Tone Method
Write a test for morseCodeTone() to ensure that the sequence for “pi” in Morse code is executed as intended

**Acceptance Criteria:**
- [ ] Unit tests correctly covers all methods
- [ ] All tests pass with mock PWM components
- [ ] Messages logged are validated for correct execution

DINES-04, 2025-04-16:
Hi @RalphTan37!
I'd love to take on this task. Could you please assign this issue to me and label it as a good first issue if that’s appropriate? I’ll make sure to cover all the required test cases with mock components and proper assertions. Looking forward to contributing!

Thanks!

DINES-04, 2025-04-16:
Hi @RalphTan37,
I hope you're doing well! I’m interested in working on this issue—would you mind assigning it to me? Also, if it’s appropriate, could you please add the good first issue label? I’d really appreciate it. Thanks a lot!

RalphTan37, 2025-04-16:
Hi @@DINES-04, 
I can absolutely complete those tasks. Thanks for the update!

DINES-04, 2025-04-22:
Hi @RalphTan37,
I just wanted to inform you that I’ve submitted a pull request for this issue,Since I had previously expressed interest and requested assignment and Please feel free to review it whenever convenient. I'm happy to make any changes if needed. 
Thank you!

DINES-04, 2025-04-22:
Hi @RalphTan37, I've submitted the PR and signed the CLA. Kindly review and approve when you get the chance. Also, the workflows are awaiting approval. Thank you!

sauravk58, 2025-09-07:
assign me